gopro/OpenGoPro #611: How can I have set_shutter enable and disable in two different files

OPEC239, 2024-09-24:
When I have the following in one file such as test.py
The gopro functions fine, start shooting and wait for 5 seconds stop shooting.

await gopro.http_command.set_shutter(shutter=Params.Toggle.ENABLE)
await asyncio.sleep(5)
await gopro.http_command.set_shutter(shutter=Params.Toggle.DISABLE)

But here comes the problem, if I put shutter enable in one file and shutter disable in another file, such as start.py and stop.py
so that I can control the shooting period only stops when I call stop.py
start.py works fine makes the gopro start shooting, but when I call stop.py gopro will not stop shooting, lookings the this file is waiting for a task to finish then starts to run. 

Can someone help me out and give me a guide, thanks

KonradIT, 2024-09-24:
Show your code for both start.py and stop.py

OPEC239, 2024-09-25:
> Show your code for both start.py and stop.py

Start.py

import asyncio
from rich.console import Console
from open_gopro import WiredGoPro, Params
console = Console()
async def main():
     try:
        async with WiredGoPro() as gopro:
            await gopro.http_command.set_shutter(shutter=Params.Toggle.ENABLE)        
     except Exception as e:
        console.print(e)
asyncio.run(main())

stop.py

import asyncio
from rich.console import Console
from open_gopro import WiredGoPro, Params
console = Console()
async def main():
     try:
        async with WiredGoPro() as gopro:
            await gopro.http_command.set_shutter(shutter=Params.Toggle.DISABLE)        
     except Exception as e:
        console.print(e)
asyncio.run(main())

tcamise-gpsw, 2025-01-07:
If I'm understanding correctly, you are simultaneously connecting to the same gopro with two different `WiredGoPro` instances.

In general, this is [not yet supported](https://github.com/gopro/OpenGoPro/issues/124).  
What is the actual benefit to doing this in your case here? It seems like it would be easier to just pass the `WiredGoPro` instance to wherever it is needed.

closed 2025-02-17 by github-actions[bot]