gopro/OpenGoPro #615: Update OpenGoPro docs

martincarapia, 2024-10-01:
Hi [martincarapia](https://github.com/martincarapia),

Hero 12 introduced support for querying livestream status over HTTP. You should be able to use the BLE endpoint for querying livestream status on Hero 9. It looks like when Hero 12 was released it looks like older camera models were removed by accident from the BLE livestream endpoint documentation. We'll try to get that fixed in the next documentation update.

_Originally posted by @bfreeland-gpsw in https://github.com/gopro/OpenGoPro/issues/609#issuecomment-2386588543_

bfreeland-gpsw, 2024-10-08:
Hi @martincarapia,

We've updated the documents to reflect the Get Livestream Status and Set Livestream Mode BLE endpoints being supported from Hero 9. We are still in the process of writing documentation for the HTTP endpoints.

martincarapia, 2024-10-08:
Wonderful, thank you for your help.

closed 2025-01-07 by tcamise-gpsw