gopro/OpenGoPro #639: GoPro in unattended - advice request

pavlosharhan2, 2024-11-23:
Hi there 👋

I've been playing with gopros for about 3 years with different devices controlling It starting from Raspberry Pis, Nvidia Jetsons, and finally Android/IOS smartphones. Throughout my path of tinkering them I had ups and downs In terms of reliability.
The main problem I was always dealing with was the GoPro entering an unrecoverable state when you need to power cycle it in order to be able to transfer/capture Clips again. 

The use case is always the same Recording 500 and more 5s clips with 20-60 sec interval with or without long breaks in between ( Imagine capturing 5sec clips in cricket game) The goal is to record 5s clip and transfer as soon as possible. Or record using bluetooth only and transfer later with wifi/usb is connected. The gopro is powered from external power source and is configured with TUSB and WAKE settings.

The problems can be categorized in few categories:

1) GoPro Wifi: GoPro Wifi disconnects and not visible anymore. restarting gopro wifi via bluetooth command may help, but not always
2) GoPro Wifi: GoPro Wifi stays connected, capturing videos without issues, but gopro/media/list returns Error 404 therefore cannot transfer clips.
3) GoPro Wifi - GoPro is recording clips, but clips on camera status field is not increasing until power cycle, and gopro/media/list returns 404 
4) GoPro Wired:  at some point GoPro starts responding with {} to start recording, and the recording doesn't start. Pressing shutter button manually doesn't start recording either. Only physical power cycle helps
5) GoPro Wired: GoPro is not visible among network interfaces on the host, and has to be reconnected/power cycled and reconnected
6) GoPro Wired: at some point gopro/media/list starts returning 404 or something like {"error_msg": "Unable to establish communication with settings controller"}. Only power cycle recovers it

I've tried investigating each specific problems and ask for advices, but It seems like the debuging is not very efficient when you don't have specific steps to repeat the problem. On some days everything works, on other day it just breaks and can't be recovered until power cycle. The only workaround i see  is adding a relay with some microcontroller that can cut the power via bluetooth command.

I would appreciate reliable advice that could help me with this, or simply knowing if something similar has been done before and how the setup and communication are structured. It would also be helpful if you could share examples of integration tests conducted in-house, so I could align my workflow more closely with those.
Additionally, if there is a way to enable debug logging on the SD card, it would be great to try that as well.

I have few hundreds of gopros laying in front of me and don't want to give up on them, I still have some hope.

If anyone had similar experience, please share it here!

Thank you.

sT0v, 2024-12-04:
@pavlosharhan2 you alluded to
> using various devices to control them—... iOS smartphones

What tools did you use to accomplish this? I am currently attempting `a-Shell, is there a more suited option?

tcamise-gpsw, 2025-01-10:
Hello. Yes, it is unfortunately hard to devote time to investigating intermittent, difficult to reproduce issues.

I can try to perform some type of similar stress testing to see if I can reproduce. Does the following seem reasonable?
1. Establish USB communication
1. Start recording a video
1. Wait 5 seconds 
1. Stop recording
1. Get the media list
1. Download the video
1. Go to step 2

Zofware, 2026-01-06:
I have a pair of Hero 10's that I use for 2 hour live streaming sessions. I hit issue #1 very frequently during those sessions. At unpredictable intervals (minutes to hours) the camera seems to lose network connectivity. The live stream drops, although recording to the SD card continues, and the camera no longer responds to pings from the network. Attempting to restart the live stream fails with the camera never able to re-establish network connectivity. Power cycling the camera clears up the issue.