gopro/OpenGoPro #645: BLE Reconnection Fails After GoPro Battery Removal or Extended Idle Time (Requires Manual Re-pairing)

areiner222, 2024-12-17:
**Component**
What is the bug in?
- Demos (python sdk, etc) - Using WirelessGoPro / python sdk

**Describe the bug**
I can successfully pair my gopro via ble and then connect with the WirelessGoPro (connecting and disconnecting successfully). However, if the battery is removed / changes or the camera sits idle for long enough after powering off then I cannot reconnect via ble without manually repairing.

**Repro**
```
gp0 = WirelessGoPro(
    target=target0,
    enable_wifi=False,
    maintain_state=False,
)
await gp0.open()

Error:
Connection failed during establishment..
[org.bluez.Error.Failed] Operation already in progress
Failed to connect: ConnectFailed('BLE connection failed to establish after 1 retries with timeout 1'). Retrying #1
Failed to connect: ConnectFailed('BLE connection failed to establish after 1 retries with timeout 1'). Retrying #2
Connection failed during establishment..
[org.bluez.Error.Failed] Operation already in progress
Error while opening: BLE connection failed to establish after 5 retries with timeout 15
Failed to connect: ConnectFailed('BLE connection failed to establish after 1 retries with timeout 1'). Retrying #3
Failed to connect: ConnectFailed('BLE connection failed to establish after 1 retries with timeout 1'). Retrying #4
---------------------------------------------------------------------------
BleakDBusError                            Traceback (most recent call last)
File [~/miniforge3/envs/zed_dev2/lib/python3.10/site-packages/bleak/__init__.py:615](http://localhost:9091/home/areiner/miniforge3/envs//lib/python3.10/site-packages/bleak/__init__.py#line=614), in BleakClient.connect(self, **kwargs)
    606 """Connect to the specified GATT server.
    607 
    608 Args:
   (...)
    613 
    614 """
--> 615 return await self._backend.connect(**kwargs)

File [~/miniforge3/envs/zed_dev2/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py:254](http://localhost:9091/home/areiner/miniforge3/envs/zed_dev2/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py#line=253), in BleakClientBlueZDBus.connect(self, dangerous_use_bleak_cache, **kwargs)
    249             raise BleakDeviceNotFoundError(
    250                 self.address,
    251                 f"Device with address {self.address} was not found. It may have been removed from BlueZ when scanning stopped.",
    252             )
--> 254     assert_reply(reply)
    256 self._is_connected = True

File [~/miniforge3/envs/zed_dev2/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py:20](http://localhost:9091/home/areiner/miniforge3/envs/zed_dev2/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py#line=19), in assert_reply(reply)
     19 if reply.message_type == MessageType.ERROR:
---> 20     raise BleakDBusError(reply.error_name, reply.body)
     21 assert reply.message_type == MessageType.METHOD_RETURN

BleakDBusError: [org.bluez.Error.Failed] Operation already in progress

The above exception was the direct cause of the following exception:

ConnectFailed                             Traceback (most recent call last)
File [~/miniforge3/envs/zed_dev2/lib/python3.10/site-packages/open_gopro/ble/client.py:110](http://localhost:9091/home/areiner/miniforge3/envs/zed_dev2/lib/python3.10/site-packages/open_gopro/ble/client.py#line=109), in BleClient.open(self, timeout, retries)
    109 try:
--> 110     self._handle = await self._controller.connect(self._disconnected_cb, self._device, timeout=timeout)
    111     break

File [~/miniforge3/envs/zed_dev2/lib/python3.10/site-packages/open_gopro/ble/adapters/bleak_wrapper.py:249](http://localhost:9091/home/areiner/miniforge3/envs/zed_dev2/lib/python3.10/site-packages/open_gopro/ble/adapters/bleak_wrapper.py#line=248), in BleakWrapperController.connect(self, disconnect_cb, device, timeout)
    248     logger.warning(exception)
--> 249     raise ConnectFailed("BLE", 1, 1) from exception
    250 return client

ConnectFailed: BLE connection failed to establish after 1 retries with timeout 1

The above exception was the direct cause of the following exception:

ConnectFailed                             Traceback (most recent call last)
[... 71 lines cut ...]
**Additional Info**

Here are some additional logs via btmon:
```
> HCI Event: Encryption Change (0x08) plen 4                                                                                                                                                                                   #453 [hci1] 39.779169
        Status: PIN or Key Missing (0x06)
        Handle: 16
        Encryption: Disabled (0x00)
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                                                                                                                                 #454 [hci1] 39.779438
        Handle: 16
        Reason: Authentication Failure (0x05)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                                                      #455 [hci1] 39.781072
      Disconnect (0x01|0x0006) ncmd 2
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                                                 #456 [hci1] 39.823092
        Status: Success (0x00)
        Handle: 16
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                                                          {0x0001} [hci1] 39.823134
        LE Address: D6:DE:15:46:94:6E (Static)
        Reason: Connection terminated due to authentication failure (0x04)
@ MGMT Event: Connect Failed (0x000d) plen 8                                                                                                                                                                               {0x0001} [hci1] 39.843463
        LE Address: D6:DE:15:46:94:6E (Static)
        Status: Disconnected (0x0e)
```

MattL-NZ, 2025-03-14:
Hey @areiner222 did you ever find a way to resolve this? I am having very similar issues.

areiner222, 2025-03-20:
@MRL-00 I never determined a very reliable way to reconnect... I have successfully connected via BLE  + COHN up to 6 cameras but I can't get it to work consistently

tcamise-gpsw, 2025-03-25:
This sounds similar to a discussion I was having in the [bleak repo](https://github.com/hbldh/bleak/discussions/1693). I'm still trying to characterize this but at least on Windows there are situations you can get into where the OS automatically reconnects and you can thus no longer connect from bleak. Also, there is no way to "access" the existing connection.

@areiner222 about your above question about establishing without BLE...unfortunately no. The Python SDK was designed before COHN existed and not in a scalable enough way to handle any type / order of connections. I've since released the Kotlin SDK which does allow such scalable connecting and plan on refactoring the Python SDK to a similar design. But this is a fairly major effort and I'm not sure when I will get to it.

tcamise-gpsw, 2025-06-26:
FYI the Python SDK has been updated to allow directory connecting COHN without BLE.

> [@MRL-00](https://github.com/MRL-00) I never determined a very reliable way to reconnect... I have successfully connected via BLE + COHN up to 6 cameras but I can't get it to work consistently
> 
> [@tcamise-gpsw](https://github.com/tcamise-gpsw) is there a way to connect to the WirelessGoPro directly via https over COHN without connecting via BLE first?

MRL-00, 2025-10-07:
@tcamise-gpsw did you ever manage to find the cause of GoPros not reconnecting if they turn off or batteries are removed from them?

We still see this happening randomly with certain devices. If we connect to 4 devices on one day, then overnight the devices power off, if we try to connect to the same 4 devices the next day, 2 of them may reconnect but 2 of them appear to fail. They have to be manually put back into pairing mode before they will reconnect.

I don't think this is a Bleak issue as we are not using the Python SDK.