gopro/OpenGoPro #648: Can't change many settings with Python SDK

mmoollllee, 2024-12-22:
**Is your feature request related to a problem or use case? Please describe.**
I've been trying to set up a GoPro Hero 13 to be managed by a RaspberryPi to remotely control the GoPro.
I'm failing to switch from HDR to Night Mode with the following code.
Other features like changing the lens mode don't work either.

```python
import asyncio
from open_gopro import WirelessGoPro, Params, proto

async def gopro():
   async with WirelessGoPro() as gopro:

      assert (await gopro.ble_command.load_preset_group(group=proto.EnumPresetGroup.PRESET_GROUP_ID_PHOTO)).ok
      print("Switched to Photo Mode")
      assert (await gopro.ble_setting.camera_ux_mode.set(Params.CameraUxMode.PRO)).ok
      print("Switched to Pro UX Mode")

      print("Photo Easy Mode Values:")
      print(await gopro.ble_setting.photo_easy_mode.get_capabilities_values())

      print("Photo Mode Values:")
      print(await gopro.ble_setting.photo_mode.get_capabilities_values())

      print("Photo Mode V2 Values:")
      print(await gopro.ble_setting.photo_mode_v2.get_capabilities_values())

      print("Trying to switch Photo Mode V2 to Night:")
      assert (await gopro.ble_setting.photo_mode_v2.set(Params.PhotoModeV2.NIGHT)).ok 

if __name__ == "__main__":
    asyncio.run(gopro())
```

returns

```
python gopro.py
Switched to Photo Mode
Switched to Pro UX Mode
Photo Easy Mode Values:
No parser defined for SettingId.INTERNAL_41
No parser defined for SettingId.INTERNAL_41
No parser defined for SettingId.INTERNAL_42
No parser defined for SettingId.INTERNAL_43
No parser defined for SettingId.INTERNAL_43
No parser defined for SettingId.INTERNAL_43
No parser defined for SettingId.INTERNAL_43
No parser defined for SettingId.INTERNAL_44
No parser defined for SettingId.INTERNAL_45
No parser defined for SettingId.INTERNAL_47
No parser defined for SettingId.INTERNAL_47
No parser defined for SettingId.INTERNAL_54
No parser defined for SettingId.INTERNAL_54
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_62
No parser defined for SettingId.INTERNAL_64
No parser defined for SettingId.INTERNAL_64
No parser defined for SettingId.INTERNAL_64
No parser defined for SettingId.INTERNAL_64
No parser defined for SettingId.INTERNAL_64
[... 914 lines cut ...]
        "SettingId.FRAME_RATE" : [
            
        ],
    },
}
Photo Mode V2 Values:
{
    "id" : "QueryCmdId.GET_CAPABILITIES_VAL",
    "status" : "ErrorCode.SUCCESS",
    "protocol" : "Protocol.BLE",
}
Received non-success status: ErrorCode.ERROR
Traceback (most recent call last):
  File "/Users/mg/dev/kuckuck-pi/gopro.py", line 38, in <module>
    asyncio.run(gopro())
  File "/Users/mg/.pyenv/versions/3.9.18/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Users/mg/.pyenv/versions/3.9.18/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/Users/mg/dev/kuckuck-pi/gopro.py", line 22, in gopro
    assert (await gopro.ble_setting.photo_mode_v2.set(Params.PhotoModeV2.NIGHT)).ok
AssertionError
```

I've been trying to follow the tutorials and demos, but after trying everything I could consider from my knowledge I'm still failing. Maybe I can get some help on this here?

tcamise-gpsw, 2025-01-07:
At a high level this is because the current setting values do not currently allow any capabilities for the Photo Mode V2 setting.
This is currently a major flaw in the API that there is no way to understand the interdependencies between settings.

But maybe there is better solution for you here. What exactly are you trying to accomplish? It is probably better to select a Preset that will achieve your desired results than settings these settings manually.

mmoollllee, 2025-01-08:
> At a high level this is because the current setting values do not currently allow any capabilities for the Photo Mode V2 setting. This is currently a major flaw in the API that there is no way to understand the interdependencies between settings.

I understand. Trying to change every other setting with impact on this I could think of, didn't help too. Is there any other way to understand these interdependencies?

> But maybe there is better solution for you here. What exactly are you trying to accomplish? It is probably better to select a Preset that will achieve your desired results than settings these settings manually.

Actually I just want to switch between HDR Mode and Night Mode for Photos and maybe add a custom ISO limit for nightmode. Can you help me with that?

tcamise-gpsw, 2025-01-08:
The general procedure here is to switch between presets, and set the HDR setting in the desired preset. Here is an example:

```
import asyncio
from pathlib import Path

from rich.console import Console

from open_gopro import Params, WirelessGoPro
from open_gopro.api.params import PresetGroup
from open_gopro.gopro_base import GoProBase
from open_gopro.logger import setup_logging
from open_gopro.proto.preset_status_pb2 import EnumPresetTitle

console = Console()

async def main() -> None:
    logger = setup_logging(__name__, Path("demo.log"))

    gopro: GoProBase | None = None

    try:
        async with WirelessGoPro(enable_wifi=False) as gopro:
            assert gopro
            # Turn off easy mode
            assert (await gopro.ble_setting.camera_ux_mode.set(Params.CameraUxMode.PRO)).ok

            # Get Presets and store desired preset IDs
            photo_presets = [
                preset
                for preset in [
                    group
                    for group in (await gopro.ble_command.get_preset_status()).data.preset_group_array
                    if group.id == PresetGroup.PHOTO
                ][0].preset_array
            ]
            photo_preset = [
                preset for preset in photo_presets if preset.title_id == EnumPresetTitle.PRESET_TITLE_PHOTO
            ][0]
            night_mode_preset = [
                preset for preset in photo_presets if preset.title_id == EnumPresetTitle.PRESET_TITLE_NIGHT
            ][0]

            console.print("Switching to photo preset")
            assert (await gopro.ble_command.load_preset(preset=photo_preset.id)).ok
            console.print("Setting HDR mode")
            assert (await gopro.ble_setting.photo_output.set(Params.PhotoOutput.HDR)).ok
            assert (await gopro.ble_command.set_shutter(shutter=Params.Toggle.ENABLE)).ok

            console.print("Switching to night photo preset")
            assert (await gopro.ble_command.load_preset(preset=night_mode_preset.id)).ok
            assert (await gopro.ble_command.set_shutter(shutter=Params.Toggle.ENABLE)).ok

    except Exception as e:  # pylint: disable = broad-except
        logger.error(repr(e))

    if gopro:
        await gopro.close()

if __name__ == "__main__":
    asyncio.run(main())
```

The Open GoPro API does not currently support ISO (or any other Protune settings)

closed 2025-02-18 by github-actions[bot]

mmoollllee, 2025-02-24:
Thank you so much for your help! This works great for WirelessGoPro.

I'm sorry to get back on this after such a long time.
Am I right, that this does not work for WiredGoPro?