gopro/OpenGoPro #649: Initial 'RequestGetPresetStatus' response throws 'InvalidProtocolBufferException'

sepp89117, 2025-01-04:
Hello everyone,

I work with Java and Android.
When parsing the initial protobuf response (Feature ID: 0xF5 and Action ID: 0xF2) from `RequestGetPresetStatus` I get an `InvalidProtocolBufferException`.
Asynchronous notification (Feature ID: 0xF5 and Action ID: 0xF3) on preset change is parsed without errors.

Aren't both responses of type `PresetStatus.NotifyPresetStatus` and parsed using the same method?

I'm testing it with `H23.01.02.32.00` and the current protocol buffers from 'https://github.com/gopro/OpenGoPro/tree/main/protobuf'.
I also tested it with different presets loaded before the request.

Thank you!

tcamise-gpsw, 2025-01-08:
Yes both initial and asynchronous responses use the same protobuf object. I just tested the following and didn't see any problems on Hero 13:
1. Factory reset
2. connect, pair
3. Register for Preset Status (and print the current status)
4. Using the camera UI, create a new Video Preset
5. Print the Preset Status update

Here is a log that includes that byte data. Can you try parsing this byte data with your generated Protobuf code to see if there are any issues. This would tell us if the problem is the Protobuf code or the data.

```
<<<<<<<<--------------
    "id" : "ActionId.GET_PRESET_STATUS",
    "feature_id" : "FeatureId.QUERY",
    "protocol" : "Protocol.BLE",
    "uuid" : "Query",
    "register_preset_status" : [
        "1"
    ],
    "unregister_preset_status" : [
        
    ],
<<<<<<<<--------------

   MainThread:11:55:58.920 gopro_wireless.py                         730 DEBUG    | Writing to [Query] UUID: 20:04:f5:72:08:01
   MainThread:11:55:58.920 bleak_wrapper.py                          101 DEBUG    | Writing to Query: b5f90076aa8d11e390460002a5d5c51b
   MainThread:11:55:59.067 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 22:52:f5:f2:0a:78:08:e8:07:12:2f:08:00:10:0c:18:0a:28:00:30
   MainThread:11:55:59.068 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 80:00:3a:07:08:e8:01:10:01:18:01:3a:06:08:02:10:01:18:01:3a
   MainThread:11:55:59.068 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 81:07:08:ea:01:10:08:18:01:3a:07:08:e5:01:10:00:18:01:40:00
   MainThread:11:55:59.069 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 82:12:38:08:01:10:20:18:6a:28:00:30:66:3a:07:08:e8:01:10:01
   MainThread:11:55:59.070 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 83:18:01:3a:06:08:02:10:64:18:01:3a:07:08:ea:01:10:01:18:01
   MainThread:11:55:59.074 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 84:3a:07:08:e5:01:10:00:18:01:3a:07:08:9c:01:10:0a:18:01:40
   MainThread:11:55:59.075 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 85:00:18:01:20:00:28:0c:28:20:28:0f:0a:81:01:08:e9:07:12:1f
   MainThread:11:55:59.076 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 86:08:80:80:04:10:10:18:03:28:00:30:03:3a:06:08:7d:10:03:18
   MainThread:11:55:59.076 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 87:01:3a:07:08:e6:01:10:1f:18:01:40:00:12:28:08:82:80:04:10
   MainThread:11:55:59.082 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 88:13:18:05:28:00:30:05:3a:06:08:7e:10:00:18:01:3a:07:08:93
   MainThread:11:55:59.082 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 89:01:10:09:18:01:3a:07:08:e7:01:10:1f:18:01:40:00:12:27:08
   MainThread:11:55:59.083 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8a:83:80:04:10:12:18:06:28:00:30:06:3a:06:08:7d:10:00:18:01
   MainThread:11:55:59.083 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8b:3a:06:08:13:10:00:18:01:3a:07:08:e6:01:10:1f:18:01:40:00
   MainThread:11:55:59.084 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8c:18:01:20:01:28:10:28:13:28:12:0a:c3:02:08:ea:07:12:30:08
   MainThread:11:55:59.089 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8d:80:80:08:10:18:18:07:28:00:30:07:3a:07:08:e8:01:10:01:18
   MainThread:11:55:59.090 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8e:01:3a:06:08:02:10:01:18:01:3a:06:08:6f:10:0a:18:01:3a:07
   MainThread:11:55:59.090 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8f:08:e5:01:10:00:18:01:40:00:12:31:08:83:80:08:10:1d:18:4c
   MainThread:11:55:59.091 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 80:28:00:30:4c:3a:07:08:e9:01:10:01:18:01:3a:06:08:02:10:01
   MainThread:11:55:59.092 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 81:18:01:3a:07:08:20:10:91:1c:18:01:3a:07:08:e5:01:10:00:18
   MainThread:11:55:59.096 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 82:01:40:00:12:31:08:84:80:08:10:1e:18:4d:28:00:30:4d:3a:07
   MainThread:11:55:59.097 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 83:08:e9:01:10:01:18:01:3a:06:08:02:10:01:18:01:3a:07:08:20
   MainThread:11:55:59.098 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 84:10:91:1c:18:01:3a:07:08:e5:01:10:00:18:01:40:00:12:31:08
   MainThread:11:55:59.098 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 85:85:80:08:10:1f:18:4e:28:00:30:4e:3a:07:08:e9:01:10:01:18
[... cut ...]
                        "id" : "131075",
                    }, {
                        "id" : "131076",
                    }, {
                        "id" : "131077",
                    }, {
                        "id" : "131073",
                    }, {
                        "id" : "131074",
                    }
                ],
                "can_add_preset" : "True",
                "icon" : "EnumPresetGroupIcon.PRESET_GROUP_TIMELAPSE_ICON_ID",
            }
        ],
    },
----ASYNC----->>>>>>>>

```

tcamise-gpsw, 2025-01-08:
I just realized that I didn't actually test your bug which if I understand, is getting the initial presets after a custom preset is created. I just did that and also saw now problems. Here is the log. This is the one you should try parsing.

```
<<<<<<<<--------------
    "id" : "ActionId.GET_PRESET_STATUS",
    "feature_id" : "FeatureId.QUERY",
    "protocol" : "Protocol.BLE",
    "uuid" : "Query",
    "register_preset_status" : [
        "1"
    ],
    "unregister_preset_status" : [
        
    ],
<<<<<<<<--------------

   MainThread:12:04:52.717 gopro_wireless.py                         730 DEBUG    | Writing to [Query] UUID: 20:04:f5:72:08:01
   MainThread:12:04:52.717 bleak_wrapper.py                          101 DEBUG    | Writing to Query: b5f90076aa8d11e390460002a5d5c51b
   MainThread:12:04:52.940 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 22:84:f5:f2:0a:a9:01:08:e8:07:12:2f:08:06:10:0c:18:16:28:01
   MainThread:12:04:52.940 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 80:30:14:3a:07:08:e8:01:10:01:18:01:3a:06:08:02:10:01:18:01
   MainThread:12:04:52.941 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 81:3a:07:08:ea:01:10:08:18:01:3a:07:08:e5:01:10:00:18:01:40
   MainThread:12:04:52.947 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 82:00:12:2f:08:00:10:0c:18:0a:28:00:30:00:3a:07:08:e8:01:10
   MainThread:12:04:52.947 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 83:01:18:01:3a:06:08:02:10:01:18:01:3a:07:08:ea:01:10:08:18
   MainThread:12:04:52.948 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 84:01:3a:07:08:e5:01:10:00:18:01:40:00:12:38:08:01:10:20:18
   MainThread:12:04:52.948 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 85:6a:28:00:30:66:3a:07:08:e8:01:10:01:18:01:3a:06:08:02:10
   MainThread:12:04:52.949 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 86:64:18:01:3a:07:08:ea:01:10:01:18:01:3a:07:08:e5:01:10:00
   MainThread:12:04:52.954 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 87:18:01:3a:07:08:9c:01:10:0a:18:01:40:00:18:01:20:00:28:0c
   MainThread:12:04:52.955 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 88:28:20:28:0f:0a:81:01:08:e9:07:12:1f:08:80:80:04:10:10:18
   MainThread:12:04:52.955 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 89:03:28:00:30:03:3a:06:08:7d:10:03:18:01:3a:07:08:e6:01:10
   MainThread:12:04:52.956 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8a:1f:18:01:40:00:12:28:08:82:80:04:10:13:18:05:28:00:30:05
   MainThread:12:04:52.957 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8b:3a:06:08:7e:10:00:18:01:3a:07:08:93:01:10:09:18:01:3a:07
   MainThread:12:04:52.961 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8c:08:e7:01:10:1f:18:01:40:00:12:27:08:83:80:04:10:12:18:06
   MainThread:12:04:52.962 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8d:28:00:30:06:3a:06:08:7d:10:00:18:01:3a:06:08:13:10:00:18
   MainThread:12:04:52.963 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8e:01:3a:07:08:e6:01:10:1f:18:01:40:00:18:01:20:01:28:10:28
   MainThread:12:04:52.963 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8f:13:28:12:0a:c3:02:08:ea:07:12:30:08:80:80:08:10:18:18:07
   MainThread:12:04:52.964 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 80:28:00:30:07:3a:07:08:e8:01:10:01:18:01:3a:06:08:02:10:01
   MainThread:12:04:52.969 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 81:18:01:3a:06:08:6f:10:0a:18:01:3a:07:08:e5:01:10:00:18:01
   MainThread:12:04:52.970 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 82:40:00:12:31:08:83:80:08:10:1d:18:4c:28:00:30:4c:3a:07:08
   MainThread:12:04:52.970 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 83:e9:01:10:01:18:01:3a:06:08:02:10:01:18:01:3a:07:08:20:10
   MainThread:12:04:52.971 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 84:91:1c:18:01:3a:07:08:e5:01:10:00:18:01:40:00:12:31:08:84
   MainThread:12:04:52.972 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 85:80:08:10:1e:18:4d:28:00:30:4d:3a:07:08:e9:01:10:01:18:01
   MainThread:12:04:52.977 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 86:3a:06:08:02:10:01:18:01:3a:07:08:20:10:91:1c:18:01:3a:07
[... cut ...]
                "can_add_preset" : "True",
                "icon" : "EnumPresetGroupIcon.PRESET_GROUP_TIMELAPSE_ICON_ID",
                "mode_array" : [
                    "EnumFlatMode.FLAT_MODE_TIME_WARP_VIDEO", "EnumFlatMode.FLAT_MODE_VIDEO_STAR_TRAIL", "EnumFlatMode.FLAT_MODE_VIDEO_LIGHT_PAINTING", "EnumFlatMode.FLAT_MODE_VIDEO_LIGHT_TRAIL", "EnumFlatMode.FLAT_MODE_TIME_LAPSE_VIDEO", "EnumFlatMode.FLAT_MODE_NIGHT_LAPSE_VIDEO"
                ],
            }
        ],
    },
-------------->>>>>>>>
```

sepp89117, 2025-01-09:
Hello and thank you for your answer!

To be clear, the steps to reproduce the bug are:

1. Factory reset
2. Create a custom preset
3. Connect, pair
4. Register for Preset Status (RequestGetPresetStatus)
5. Parse the initial response of RequestGetPresetStatus

I tested both of the (initial) response data from your logs.

The first one without error. 
> Here is a log that includes that byte data. Can you try parsing this byte data with your generated Protobuf code to see if there are any issues. This would tell us if the problem is the Protobuf code or the data.
```java
byte[] unproblematicProtoResp = {0xf5, 0xf2, 0x0a, 0x78, 0x08, 0xe8, 0x07, 0x12, 0x2f, 0x08, 0x00, 0x10, 0x0c, 0x18, 0x0a, 0x28, 0x00, 0x30, 0x00, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xea, 0x01, 0x10, 0x08, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x38, 0x08, 0x01, 0x10, 0x20, 0x18, 0x6a, 0x28, 0x00, 0x30, 0x66, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x64, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xea, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x9c, 0x01, 0x10, 0x0a, 0x18, 0x01, 0x40, 0x00, 0x18, 0x01, 0x20, 0x00, 0x28, 0x0c, 0x28, 0x20, 0x28, 0x0f, 0x0a, 0x81, 0x01, 0x08, 0xe9, 0x07, 0x12, 0x1f, 0x08, 0x80, 0x80, 0x04, 0x10, 0x10, 0x18, 0x03, 0x28, 0x00, 0x30, 0x03, 0x3a, 0x06, 0x08, 0x7d, 0x10, 0x03, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe6, 0x01, 0x10, 0x1f, 0x18, 0x01, 0x40, 0x00, 0x12, 0x28, 0x08, 0x82, 0x80, 0x04, 0x10, 0x13, 0x18, 0x05, 0x28, 0x00, 0x30, 0x05, 0x3a, 0x06, 0x08, 0x7e, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x93, 0x01, 0x10, 0x09, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe7, 0x01, 0x10, 0x1f, 0x18, 0x01, 0x40, 0x00, 0x12, 0x27, 0x08, 0x83, 0x80, 0x04, 0x10, 0x12, 0x18, 0x06, 0x28, 0x00, 0x30, 0x06, 0x3a, 0x06, 0x08, 0x7d, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x13, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe6, 0x01, 0x10, 0x1f, 0x18, 0x01, 0x40, 0x00, 0x18, 0x01, 0x20, 0x01, 0x28, 0x10, 0x28, 0x13, 0x28, 0x12, 0x0a, 0xc3, 0x02, 0x08, 0xea, 0x07, 0x12, 0x30, 0x08, 0x80, 0x80, 0x08, 0x10, 0x18, 0x18, 0x07, 0x28, 0x00, 0x30, 0x07, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x6f, 0x10, 0x0a, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x31, 0x08, 0x83, 0x80, 0x08, 0x10, 0x1d, 0x18, 0x4c, 0x28, 0x00, 0x30, 0x4c, 0x3a, 0x07, 0x08, 0xe9, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x20, 0x10, 0x91, 0x1c, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x31, 0x08, 0x84, 0x80, 0x08, 0x10, 0x1e, 0x18, 0x4d, 0x28, 0x00, 0x30, 0x4d, 0x3a, 0x07, 0x08, 0xe9, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x20, 0x10, 0x91, 0x1c, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x31, 0x08, 0x85, 0x80, 0x08, 0x10, 0x1f, 0x18, 0x4e, 0x28, 0x00, 0x30, 0x4e, 0x3a, 0x07, 0x08, 0xe9, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x20, 0x10, 0x91, 0x1c, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x30, 0x08, 0x81, 0x80, 0x08, 0x10, 0x0d, 0x18, 0x08, 0x28, 0x00, 0x30, 0x08, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x05, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x31, 0x08, 0x82, 0x80, 0x08, 0x10, 0x1a, 0x18, 0x09, 0x28, 0x00, 0x30, 0x09, 0x3a, 0x07, 0x08, 0xe9, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x20, 0x10, 0x91, 0x1c, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x18, 0x01, 0x20, 0x02, 0x28, 0x18, 0x28, 0x1d, 0x28, 0x1e, 0x28, 0x1f, 0x28, 0x0d, 0x28, 0x1a, 0x12, 0x04, 0x08, 0x12, 0x10, 0x21, 0x1a, 0x04, 0x08, 0x13, 0x10, 0x20};
```

The second one with the same error `com.google.protobuf.InvalidProtocolBufferException: Protocol message contained an invalid tag (zero).`
> I just realized that I didn't actually test your bug which if I understand, is getting the initial presets after a custom preset is created. I just did that and also saw now problems. Here is the log. This is the one you should try parsing.
```java
byte[] problematicProtoResp = {0xf5, 0xf2, 0x0a, 0xa9, 0x01, 0x08, 0xe8, 0x07, 0x12, 0x2f, 0x08, 0x06, 0x10, 0x0c, 0x18, 0x16, 0x28, 0x01, 0x30, 0x14, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xea, 0x01, 0x10, 0x08, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x2f, 0x08, 0x00, 0x10, 0x0c, 0x18, 0x0a, 0x28, 0x00, 0x30, 0x00, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xea, 0x01, 0x10, 0x08, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x38, 0x08, 0x01, 0x10, 0x20, 0x18, 0x6a, 0x28, 0x00, 0x30, 0x66, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x64, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xea, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0 [... cut ...] 0x1a, 0x18, 0x09, 0x28, 0x00, 0x30, 0x09, 0x3a, 0x07, 0x08, 0xe9, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x20, 0x10, 0x91, 0x1c, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x18, 0x01, 0x20, 0x02, 0x28, 0x18, 0x28, 0x1d, 0x28, 0x1e, 0x28, 0x1f, 0x28, 0x0d, 0x28, 0x1a, 0x12, 0x04, 0x08, 0x12, 0x10, 0x21, 0x1a, 0x04, 0x08, 0x13, 0x10, 0x20};
```
But as I understand it, you also had problems with that, right?

tcamise-gpsw, 2025-01-09:
Sorry I mistyped above. To be clear: I did not see any problems parsing any of the data above. I took the logs from the Python SDK.

Anyway it is good that we now have a test vector (`problematicProtoResp`) to replicate the issue from your perspective.

Now here is where it gets weird...I just tried to parse the `problematicProtoResp` directly using the structures from the Python SDK and I do actually see a similar error now: `Field number 0 is illegal.`

I can't explain this but will do some debugging.

sepp89117, 2025-01-10:
Okay, that sounds really strange at first. But it's the data from your log that I should test.
```
   MainThread:12:04:52.940 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 22:84:f5:f2:0a:a9:01:08:e8:07:12:2f:08:06:10:0c:18:16:28:01
   MainThread:12:04:52.940 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 80:30:14:3a:07:08:e8:01:10:01:18:01:3a:06:08:02:10:01:18:01
   MainThread:12:04:52.941 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 81:3a:07:08:ea:01:10:08:18:01:3a:07:08:e5:01:10:00:18:01:40
   MainThread:12:04:52.947 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 82:00:12:2f:08:00:10:0c:18:0a:28:00:30:00:3a:07:08:e8:01:10
   MainThread:12:04:52.947 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 83:01:18:01:3a:06:08:02:10:01:18:01:3a:07:08:ea:01:10:08:18
   MainThread:12:04:52.948 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 84:01:3a:07:08:e5:01:10:00:18:01:40:00:12:38:08:01:10:20:18
   MainThread:12:04:52.948 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 85:6a:28:00:30:66:3a:07:08:e8:01:10:01:18:01:3a:06:08:02:10
   MainThread:12:04:52.949 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 86:64:18:01:3a:07:08:ea:01:10:01:18:01:3a:07:08:e5:01:10:00
   MainThread:12:04:52.954 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 87:18:01:3a:07:08:9c:01:10:0a:18:01:40:00:18:01:20:00:28:0c
   MainThread:12:04:52.955 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 88:28:20:28:0f:0a:81:01:08:e9:07:12:1f:08:80:80:04:10:10:18
   MainThread:12:04:52.955 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 89:03:28:00:30:03:3a:06:08:7d:10:03:18:01:3a:07:08:e6:01:10
   MainThread:12:04:52.956 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8a:1f:18:01:40:00:12:28:08:82:80:04:10:13:18:05:28:00:30:05
   MainThread:12:04:52.957 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8b:3a:06:08:7e:10:00:18:01:3a:07:08:93:01:10:09:18:01:3a:07
   MainThread:12:04:52.961 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8c:08:e7:01:10:1f:18:01:40:00:12:27:08:83:80:04:10:12:18:06
   MainThread:12:04:52.962 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8d:28:00:30:06:3a:06:08:7d:10:00:18:01:3a:06:08:13:10:00:18
   MainThread:12:04:52.963 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8e:01:3a:07:08:e6:01:10:1f:18:01:40:00:18:01:20:01:28:10:28
   MainThread:12:04:52.963 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8f:13:28:12:0a:c3:02:08:ea:07:12:30:08:80:80:08:10:18:18:07
   MainThread:12:04:52.964 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 80:28:00:30:07:3a:07:08:e8:01:10:01:18:01:3a:06:08:02:10:01
   MainThread:12:04:52.969 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 81:18:01:3a:06:08:6f:10:0a:18:01:3a:07:08:e5:01:10:00:18:01
   MainThread:12:04:52.970 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 82:40:00:12:31:08:83:80:08:10:1d:18:4c:28:00:30:4c:3a:07:08
   MainThread:12:04:52.970 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 83:e9:01:10:01:18:01:3a:06:08:02:10:01:18:01:3a:07:08:20:10
   MainThread:12:04:52.971 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 84:91:1c:18:01:3a:07:08:e5:01:10:00:18:01:40:00:12:31:08:84
   MainThread:12:04:52.972 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 85:80:08:10:1e:18:4d:28:00:30:4d:3a:07:08:e9:01:10:01:18:01
   MainThread:12:04:52.977 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 86:3a:06:08:02:10:01:18:01:3a:07:08:20:10:91:1c:18:01:3a:07
   MainThread:12:04:52.977 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 87:08:e5:01:10:00:18:01:40:00:12:31:08:85:80:08:10:1f:18:4e
   MainThread:12:04:52.978 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 88:28:00:30:4e:3a:07:08:e9:01:10:01:18:01:3a:06:08:02:10:01
   MainThread:12:04:52.979 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 89:18:01:3a:07:08:20:10:91:1c:18:01:3a:07:08:e5:01:10:00:18
   MainThread:12:04:52.980 gopro_wireless.py                         684 DEBUG    | Received response on BleUUID [Query Response]: 8a:01:40:00:12:30:08:81:80:08:10:0d:18:08:28:00:30:08:3a:07
[... cut ...]
4. Remove `84, ` from the beginning of the payload. (Removes the rest of the first package header)
In this way I get a payload of the received data to parse, as in my program

The only logical explanation for me is that your software changes the data before parsing it. So I checked your code and saw  [here](https://github.com/gopro/OpenGoPro/blob/main/demos/python/sdk_wireless_camera_control/open_gopro/models/response.py#L446) that you remove the FeatureID from the payload. I will run the data without the FeatureID through the Proto-PresetStatus parser and see what happens.

sepp89117, 2025-01-10:
Without the FeatureID I was now able to parse both payloads without errors.

```java
byte[] unproblematicProtoResp = {0xf2, 0x0a, 0x78, 0x08, 0xe8, 0x07, 0x12, 0x2f, 0x08, 0x00, 0x10, 0x0c, 0x18, 0x0a, 0x28, 0x00, 0x30, 0x00, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xea, 0x01, 0x10, 0x08, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x38, 0x08, 0x01, 0x10, 0x20, 0x18, 0x6a, 0x28, 0x00, 0x30, 0x66, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x64, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xea, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x9c, 0x01, 0x10, 0x0a, 0x18, 0x01, 0x40, 0x00, 0x18, 0x01, 0x20, 0x00, 0x28, 0x0c, 0x28, 0x20, 0x28, 0x0f, 0x0a, 0x81, 0x01, 0x08, 0xe9, 0x07, 0x12, 0x1f, 0x08, 0x80, 0x80, 0x04, 0x10, 0x10, 0x18, 0x03, 0x28, 0x00, 0x30, 0x03, 0x3a, 0x06, 0x08, 0x7d, 0x10, 0x03, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe6, 0x01, 0x10, 0x1f, 0x18, 0x01, 0x40, 0x00, 0x12, 0x28, 0x08, 0x82, 0x80, 0x04, 0x10, 0x13, 0x18, 0x05, 0x28, 0x00, 0x30, 0x05, 0x3a, 0x06, 0x08, 0x7e, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x93, 0x01, 0x10, 0x09, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe7, 0x01, 0x10, 0x1f, 0x18, 0x01, 0x40, 0x00, 0x12, 0x27, 0x08, 0x83, 0x80, 0x04, 0x10, 0x12, 0x18, 0x06, 0x28, 0x00, 0x30, 0x06, 0x3a, 0x06, 0x08, 0x7d, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x13, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe6, 0x01, 0x10, 0x1f, 0x18, 0x01, 0x40, 0x00, 0x18, 0x01, 0x20, 0x01, 0x28, 0x10, 0x28, 0x13, 0x28, 0x12, 0x0a, 0xc3, 0x02, 0x08, 0xea, 0x07, 0x12, 0x30, 0x08, 0x80, 0x80, 0x08, 0x10, 0x18, 0x18, 0x07, 0x28, 0x00, 0x30, 0x07, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x6f, 0x10, 0x0a, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x31, 0x08, 0x83, 0x80, 0x08, 0x10, 0x1d, 0x18, 0x4c, 0x28, 0x00, 0x30, 0x4c, 0x3a, 0x07, 0x08, 0xe9, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x20, 0x10, 0x91, 0x1c, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x31, 0x08, 0x84, 0x80, 0x08, 0x10, 0x1e, 0x18, 0x4d, 0x28, 0x00, 0x30, 0x4d, 0x3a, 0x07, 0x08, 0xe9, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x20, 0x10, 0x91, 0x1c, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x31, 0x08, 0x85, 0x80, 0x08, 0x10, 0x1f, 0x18, 0x4e, 0x28, 0x00, 0x30, 0x4e, 0x3a, 0x07, 0x08, 0xe9, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x20, 0x10, 0x91, 0x1c, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x30, 0x08, 0x81, 0x80, 0x08, 0x10, 0x0d, 0x18, 0x08, 0x28, 0x00, 0x30, 0x08, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x05, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x31, 0x08, 0x82, 0x80, 0x08, 0x10, 0x1a, 0x18, 0x09, 0x28, 0x00, 0x30, 0x09, 0x3a, 0x07, 0x08, 0xe9, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x20, 0x10, 0x91, 0x1c, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x18, 0x01, 0x20, 0x02, 0x28, 0x18, 0x28, 0x1d, 0x28, 0x1e, 0x28, 0x1f, 0x28, 0x0d, 0x28, 0x1a, 0x12, 0x04, 0x08, 0x12, 0x10, 0x21, 0x1a, 0x04, 0x08, 0x13, 0x10, 0x20};
```
and
```java
byte[] nowUnproblematicProtoResp = {0xf2, 0x0a, 0xa9, 0x01, 0x08, 0xe8, 0x07, 0x12, 0x2f, 0x08, 0x06, 0x10, 0x0c, 0x18, 0x16, 0x28, 0x01, 0x30, 0x14, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xea, 0x01, 0x10, 0x08, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x2f, 0x08, 0x00, 0x10, 0x0c, 0x18, 0x0a, 0x28, 0x00, 0x30, 0x00, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xea, 0x01, 0x10, 0x08, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x12, 0x38, 0x08, 0x01, 0x10, 0x20, 0x18, 0x6a, 0x28, 0x00, 0x30, 0x66, 0x3a, 0x07, 0x08, 0xe8, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x64, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xea, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x9c, 0x01, 0x10, 0x0a, 0x18, 0x01, 0x40, 0x00, 0x18, 0x01, 0x20, 0x00, 0x28, 0x0c, 0x28, 0x20, 0x28, 0x0f, 0x0a, 0x81, 0x01, 0x08, 0xe9, 0x07, 0x12, 0x1f, 0x08, 0x80, 0x80, 0x04, 0x10, 0x10, 0x18, 0x03, 0x28, 0x00, 0x30, 0x03, 0x3a, 0x06, 0x08, 0x7d, 0x10, 0x03, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe6, 0x01, 0x10, 0x1f, 0x18, 0x01, 0x40, 0x00, 0x12, 0x28, 0x08, 0x82, 0x80, 0x04, 0x10, 0x13, 0x18, 0x05, 0x28, 0x00, 0x30, 0x05, 0x3a, 0x06, 0x08, 0x7e, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x93, 0x01, 0x10, 0x09, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe7, 0x01, 0x10, 0x1f, 0x18, 0x01, 0x40, 0x00, 0x12, 0x27, 0x08, 0x83, 0x80, 0x04, 0x10, 0x12, 0x18, 0x06, 0x28, 0x00, 0x30, 0x06, 0x3a, 0x06, 0x08, 0x7d, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x13, 0x10, 0x00, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe6, 0x01, 0x10, 0x1f, 0x18, 0x01, 0x40, 0x00, 0 [... cut ...] 8, 0x82, 0x80, 0x08, 0x10, 0x1a, 0x18, 0x09, 0x28, 0x00, 0x30, 0x09, 0x3a, 0x07, 0x08, 0xe9, 0x01, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x06, 0x08, 0x02, 0x10, 0x01, 0x18, 0x01, 0x3a, 0x07, 0x08, 0x20, 0x10, 0x91, 0x1c, 0x18, 0x01, 0x3a, 0x07, 0x08, 0xe5, 0x01, 0x10, 0x00, 0x18, 0x01, 0x40, 0x00, 0x18, 0x01, 0x20, 0x02, 0x28, 0x18, 0x28, 0x1d, 0x28, 0x1e, 0x28, 0x1f, 0x28, 0x0d, 0x28, 0x1a, 0x12, 0x04, 0x08, 0x12, 0x10, 0x21, 0x1a, 0x04, 0x08, 0x13, 0x10, 0x20};
```

I think we can close the issue now.

sepp89117, 2025-01-14:
I think the issue can be closed. Thanks for the support

closed 2025-01-14 by sepp89117