gopro/OpenGoPro #656: BLE pairing without connections menu selection each time?

kmatzen, 2025-01-12:
I have a python script that uses BleakClient to query my GoPro Hero 10's, connect and pair with each, and then write some settings to all.  However, I can discover the BLE addresses easily, but for them to connect, I need to follow the instructions to bring up the quik pairing screen each time I want to connect.  The quik app on iOS seems to be able to connect every time without needing the connections menu to be used each time.  Is there a way that I can do this with my own script?  I don't want to have to swipe through some menus on each camera several times just to get them to connect.

Also, I seem to be able to get the connection to happen regardless of whether I select "gopro quik app", "the remote", "volta", or "waterproof shutter remote" in the ble connections menu.  How do these work?  Is there some registration data that I'm supposed to provide in a ble request before disconnecting so that the gopro knows to accept the connection without the UI interactions?  Is there a way to make a custom app for this list?

If the answer is no, I guess I can switch to using wifi to bootstrap configuring a bunch of gopros, but I was hoping to save battery with ble.

kmatzen, 2025-01-12:
I recorded some logs and found that the behavior depends on the device.
* iPhone - I used nRF Connect and it was able to connect to the GoPro whether the quik pairing screen on the camera was open or not.  
* MBP - I used BlueSee and was able to connect if the quik pairing screen on the camera was open, but could not connect if it was not open.

After reading the logs, bluetoothd on both devices seems to do similar things until right after the GATT device is connected.  On iOS encryption is enabled.  On macOS when the quik menu is open, encryption is enabled.  On macOS when the quik menu is not open, There's a _GATT_LE_DisconnectedCB event with status 762. Then there's a notification that "Encryption failed due to  STATUS 634".

Is it the case that Apple might have some out of band method implemented for iOS that can wake the accessory so that some sort of handshake can succeed?

tcamise-gpsw, 2025-03-25:
Hmmm. This is supposed to be fixed and I wasn't seeing but admittedly haven't tested on Mac in a while. 
I'll need to do some more testing but anyway let's keep this discussion in the original #597

closed 2025-03-25 by tcamise-gpsw