gopro/OpenGoPro #666: Cannot connect to my GoPro HERO 4K (2024) via USB nor WiFi.

krzywiecki, 2025-02-15:
Hi guys!

I'm trying to connect my GoPro HERO 4K (2024) to a Raspberry Pi 4 and use it as a Webcam. I followed the instructions from the documentation but it doesn't work, my Raspberry Pi is not detecting the camera at all (tried using both wired USB and WiFi connections). I managed to do it with my Hero 11 but other options are not supported so I got new HERO. Also, the size of the camera plays crucial role for my use case so I'd love to stick to the HERO. Can you please advise on how to get it solved?

KonradIT, 2025-02-15:
HERO 2024 is not supported in OpenGoPro and also doesn't support webcam functionality.

krzywiecki, 2025-02-15:
Yeah, I figured that and wondering what would be the simplest workaround for that, it's the same firmware in the end, right?

Alternatively, what would be a good choice of a camera considering that I'd appreciate a similar size of the device? GoPro 11 mini?

KonradIT, 2025-02-15:
11 Mini doesn't do webcam. Only Hero8-13 Black gen cameras do USB livestreaming (webcam). You could take one apart and make it naked, if you're powering from USB to get the webcam output, you don't need the battery in there.

krzywiecki, 2025-02-16:
Thanks for the answer Konrad! Indeed, making it naked is something I'm considering now.

tcamise-gpsw, 2025-03-19:
closing since Hero 2024 is not [supported ](https://gopro.github.io/OpenGoPro/ble/#supported-cameras)as mentioned above

closed 2025-03-19 by tcamise-gpsw