gopro/OpenGoPro #685: Gopro12 timeout error when downloading media to host via USB

robertobaratheon, 2025-03-14:
What is the bug in?
- Demos (python sdk, etc)

I'm testing the 'video.py' demo.
I set it to record, say, for 10 seconds (I've tried multiple durations), and the script successfully exits automatically after downloading the video.
After a few loops, it will crash with:

`ERROR    12:18:36.832 ConnectionError(ReadTimeoutError("HTTPConnectionPool(host='172.27.154.51', port=8080): Read timed out."))`

This is random, it can happen after 2 or 3 loops, or after 3+ hours of loops without crashing.
When adding more logging for debugging, I've even found out this error:

```
WARNING  17:47:12.624 ConnectionError(MaxRetryError("HTTPConnectionPool(host='172.27.154.51', port=8080): Max retries exceeded with url: /gopro/camera/control/wired_usb?p=1 (Caused by            gopro_base.py:391
NewConnectionError('<urllib3.connection.HTTPConnection object at 0x1055a0ec0>: Failed to establish a new connection: [Errno 64] Host is down'))"))
```

These are very small video files I'm testing with, since the camera is static, shooting at 1080p. They are usually less than 2MB in size. Generally speaking they will download in less than a second.

I also found that when it times out, it's always after having already downloaded more than 90% of the file.

This is a full error output:

```
INFO     12:18:31.688                                                                                                                                                                              gopro_base.py:380
         <<<<<<<<--------------
             "id" : "Media List",
             "protocol" : "Protocol.HTTP",
             "endpoint" : "gopro/media/list",
         <<<<<<<<--------------

INFO     12:18:31.706                                                                                                                                                                              gopro_base.py:400
         -------------->>>>>>>>
             "id" : "http://172.27.154.51:8080/gopro/media/list",
             "status" : "ErrorCode.SUCCESS",
             "protocol" : "Protocol.HTTP",
             "data" : {
                 "identifier" : "2391561004809514755",
                 "media" : [
                     {
                         "directory" : "100GOPRO",
                         "file_system" : [
                             {
                                 "filename" : "100GOPRO/GX011325.MP4",
                                 "creation_timestamp" : "1741954696",
                                 "modified_time" : "1741954696",
                                 "low_res_video_size" : "1130887",
                                 "lrv_file_size" : "-1",
                             }
                         ],
                     }
                 ],
             },
         -------------->>>>>>>>

Downloading 100GOPRO/GX011325.MP4...
INFO     12:18:31.708                                                                                                                                                                              gopro_base.py:380
         <<<<<<<<--------------
             "id" : "Camera State",
             "protocol" : "Protocol.HTTP",
             "endpoint" : "gopro/camera/state",
         <<<<<<<<--------------

INFO     12:18:31.751                                                                                                                                                                              gopro_base.py:400
         -------------->>>>>>>>
             "id" : "http://172.27.154.51:8080/gopro/camera/state",
             "status" : "ErrorCode.SUCCESS",
             "protocol" : "Protocol.HTTP",
             "data" : {
                 "StatusId.BATTERY_PRESENT" : "False",
[... 105 lines cut ...]
                 "SettingId.FRAMING" : "Framing.WIDESCREEN",
             },
         -------------->>>>>>>>

ERROR    12:18:36.832 ConnectionError(ReadTimeoutError("HTTPConnectionPool(host='172.27.154.51', port=8080): Read timed out."))                                                                          video.py:68
```

I've recreated the error in 2 different Gopro 12 cameras.

Steps to reproduce the behavior:
I run the script with this loop:
```
python3 video.py --wired -r 10
  if [ $? -ne 0 ]; then
    echo "Python script crashed. Exiting loop."
    break
  fi
  echo 'sleeping for 3'
  sleep 3
done
```

Hardware
2 different Gopro 12 cameras.
Tested on macOS Sequoia 15.3.1 and 2 different Raspberry pi 5

viktor76525, 2025-03-20:
I think I'm experiencing the same thing. The GoPro randomly stops responding to requests over USB network.
Here's an example script (dependencies:  curl, ip, jq in path). It's meant to take photos in a loop, download them and remove them from the camera. The GoPro will continuously take periods of rest where it won't respond to anything.
[gopro2.sh.txt](https://github.com/user-attachments/files/19358487/gopro2.sh.txt)

isa8ell4, 2025-03-25:
I am having a similar issue. I have two laptops, two gopro hero 13 cameras. One laptop works with the CLI commands below and the other does not. Same versions of open-gopro (0.19.4) and python. I am controlling the gopros via CLI. Here are some of the responses from the laptop that is having issues: 

CLI: gopro-photo --wired

INFO     09:02:46.093 Querying mDNS to find _gopro-web._tcp.local....                                mdns_scanner.py:69
ERROR    09:02:56.109 FailedToFindDevice('A scan timed out without finding a device')                       photo.py:51

CLI: gopro-photo --output image.jpg --wired --identifier .*7576 --log 7576.txt
 

INFO     09:15:05.829                                                                                 gopro_base.py:380
         <<<<<<<<--------------
             "id" : "Camera Control Wired Usb",
             "protocol" : "Protocol.HTTP",
             "endpoint" : "gopro/camera/control/wired_usb",
             "p" : "Toggle.ENABLE",
         <<<<<<<<--------------

WARNING  09:15:10.859 ConnectTimeout(MaxRetryError("HTTPConnectionPool(host='172.25.176.51',          gopro_base.py:391
         port=8080): Max retries exceeded with url: /gopro/camera/control/wired_usb?p=1 (Caused by
         ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x0000020A8BBCCEF0>,
         'Connection to 172.25.176.51 timed out. (connect timeout=5)'))"))
WARNING  09:15:12.874 Retrying #1 to send the command...                                              gopro_base.py:396
WARNING  09:15:17.888 ConnectTimeout(MaxRetryError("HTTPConnectionPool(host='172.25.176.51',          gopro_base.py:391
         port=8080): Max retries exceeded with url: /gopro/camera/control/wired_usb?p=1 (Caused by
         ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x0000020A8BBCE030>,
         'Connection to 172.25.176.51 timed out. (connect timeout=5)'))"))
WARNING  09:15:19.903 Retrying #2 to send the command...                                              gopro_base.py:396
WARNING  09:15:24.912 ConnectTimeout(MaxRetryError("HTTPConnectionPool(host='172.25.176.51',          gopro_base.py:391
         port=8080): Max retries exceeded with url: /gopro/camera/control/wired_usb?p=1 (Caused by
         ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x0000020A8BBCECC0>,
         'Connection to 172.25.176.51 timed out. (connect timeout=5)'))"))
WARNING  09:15:26.927 Retrying #3 to send the command...                                              gopro_base.py:396
WARNING  09:15:31.929 ConnectTimeout(MaxRetryError("HTTPConnectionPool(host='172.25.176.51',          gopro_base.py:391
         port=8080): Max retries exceeded with url: /gopro/camera/control/wired_usb?p=1 (Caused by
         ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x0000020A8BBCF890>,
         'Connection to 172.25.176.51 timed out. (connect timeout=5)'))"))
WARNING  09:15:33.944 Retrying #4 to send the command...                                              gopro_base.py:396
WARNING  09:15:38.959 ConnectTimeout(MaxRetryError("HTTPConnectionPool(host='172.25.176.51',          gopro_base.py:391
         port=8080): Max retries exceeded with url: /gopro/camera/control/wired_usb?p=1 (Caused by
         ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x0000020A8BBCD820>,
         'Connection to 172.25.176.51 timed out. (connect timeout=5)'))"))
WARNING  09:15:40.974 Retrying #5 to send the command...                                              gopro_base.py:396
ERROR    09:15:40.974 ResponseTimeout('Response timeout occurred of 5 seconds')                             photo.py:51

This happens immediately as opposed to running it for hours. If anyone has any debugging suggestions I am all-ears :)

tcamise-gpsw, 2025-03-25:
> I am having a similar issue. I have two laptops, two gopro hero 13 cameras. One laptop works with the CLI commands below and the other does not. Same versions of open-gopro (0.19.4) and python. I am controlling the gopros via CLI. Here are some of the responses from the laptop that is having issues:
> 
> CLI: gopro-photo --wired
> 
> INFO 09:02:46.093 Querying mDNS to find _gopro-web._tcp.local.... mdns_scanner.py:69 ERROR 09:02:56.109 FailedToFindDevice('A scan timed out without finding a device') photo.py:51
> 
> CLI: gopro-photo --output image.jpg --wired --identifier .*7576 --log 7576.txt
> 
> INFO 09:15:05.829 gopro_base.py:380 <<<<<<<<-------------- "id" : "Camera Control Wired Usb", "protocol" : "Protocol.HTTP", "endpoint" : "gopro/camera/control/wired_usb", "p" : "Toggle.ENABLE", <<<<<<<<--------------
> 
> WARNING 09:15:10.859 ConnectTimeout(MaxRetryError("HTTPConnectionPool(host='172.25.176.51', gopro_base.py:391 port=8080): Max retries exceeded with url: /gopro/camera/control/wired_usb?p=1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x0000020A8BBCCEF0>, 'Connection to 172.25.176.51 timed out. (connect timeout=5)'))")) WARNING 09:15:12.874 Retrying [#1](https://github.com/gopro/OpenGoPro/pull/1) to send the command... gopro_base.py:396 WARNING 09:15:17.888 ConnectTimeout(MaxRetryError("HTTPConnectionPool(host='172.25.176.51', gopro_base.py:391 port=8080): Max retries exceeded with url: /gopro/camera/control/wired_usb?p=1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x0000020A8BBCE030>, 'Connection to 172.25.176.51 timed out. (connect timeout=5)'))")) WARNING 09:15:19.903 Retrying [#2](https://github.com/gopro/OpenGoPro/pull/2) to send the command... gopro_base.py:396 WARNING 09:15:24.912 ConnectTimeout(MaxRetryError("HTTPConnectionPool(host='172.25.176.51', gopro_base.py:391 port=8080): Max retries exceeded with url: /gopro/camera/control/wired_usb?p=1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x0000020A8BBCECC0>, 'Connection to 172.25.176.51 timed out. (connect timeout=5)'))")) WARNING 09:15:26.927 Retrying [#3](https://github.com/gopro/OpenGoPro/pull/3) to send the command... gopro_base.py:396 WARNING 09:15:31.929 ConnectTimeout(MaxRetryError("HTTPConnectionPool(host='172.25.176.51', gopro_base.py:391 port=8080): Max retries exceeded with url: /gopro/camera/control/wired_usb?p=1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x0000020A8BBCF890>, 'Connection to 172.25.176.51 timed out. (connect timeout=5)'))")) WARNING 09:15:33.944 Retrying [#4](https://github.com/gopro/OpenGoPro/pull/4) to send the command... gopro_base.py:396 WARNING 09:15:38.959 ConnectTimeout(MaxRetryError("HTTPConnectionPool(host='172.25.176.51', gopro_base.py:391 port=8080): Max retries exceeded with url: /gopro/camera/control/wired_usb?p=1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x0000020A8BBCD820>, 'Connection to 172.25.176.51 timed out. (connect timeout=5)'))")) WARNING 09:15:40.974 Retrying [#5](https://github.com/gopro/OpenGoPro/pull/5) to send the command... gopro_base.py:396 ERROR 09:15:40.974 ResponseTimeout('Response timeout occurred of 5 seconds') photo.py:51
> 
> This happens immediately as opposed to running it for hours. If anyone has any debugging suggestions I am all-ears :)
> 
> EDIT / SOLUTION: Turns out I did not have the drivers installed on the second laptop. On the first laptop I had downloaded the GoPro Webcam Utility for something else a couple weeks ago, which had installed the drivers for my laptop to communicate with the GoPro.

We do have that mentioned in the [docs ](https://gopro.github.io/OpenGoPro/python_sdk/installation.html#external-dependencies)but its definitely understandable that you missed it. Unfortunately I'm not sure how to improve visibility here since I have no way of knowing programmatically whether the drivers are installed.

tcamise-gpsw, 2025-03-25:
In regards to the original ticket, I think this all just falls under the fact that the camera is not currently robustly tested for this type of continuous operation. See also:
- #639 
- #592 
- #519  
- #184 

I'll try to carve out some time to perform this type of testing

robertobaratheon, 2025-04-03:
Hi @tcamise-gpsw 
Thanks for the reply. I understand what you mean about robustness.
I have to say, however, that this issue doesn't only happen when testing on a loop continuously.
Sometimes it happens straight away, on the very first try.
I posted the script with the loop I was testing, so that it would be easy to reproduce the error, since sometimes trying a single time or a couple of times, it wouldn't manifest.
But as I say, sometimes, the timeout error will happen straight away.
If I would have to guess, I would say it could potentially be a firmware bug where the camera server times out?
But I'm just talking without really knowing enough here.

tcamise-gpsw, 2025-04-17:
I'm kind of confused here. I would expect the following to always work:
1. Pull battery
2. Put battery back in
3. Run video script

If this does fail, then there might be some issue where the camera's HTTP server is not ready after connecting. I have seen this but the SDK should handle this since it continuously retries. I've also seen the retries eventually work.,

Any other type of continuous reconnection / redownloading while the camera has not be hard reset is going to fall under the robustness investigation.

robertobaratheon, 2025-04-23:
@tcamise-gpsw 
This is not using a battery, but powering the camera via USB.
I'm using the Gopro Labs firmware, with USB power trust override and Wake on Power.