gopro/OpenGoPro #686: Wi-Fi remote pairing through OpenGopro problem

arodrigoca, 2025-03-19:
Hello,

I have a GoPro Hero 9 with non-functioning rear screen. I would like to know if it is possible to set the camera to be on remote pairing mode through the BLE API.
As far as I know, the GoPro original remote creates a Wi-Fi access point (only discoverable in the remote's pairing mode). Through the OpenGopro turorial 6, I managed to make the camera connect as a station to the Wi-Fi remote access point. However, after a few seconds, the camera disconnects from the remote.

The tutorial 6 example code reports some problem with provisioning:

`INFO     17:26:01.178 Connecting to HERO-RC-[REDACTED-REMOTE-ID]                                                                                      connect_as_sta.py:160DEBUG    17:26:01.184 Writing: 02:04:0a:0e:48:45:52:4f:2d:52:43:2d:34:42:35:34:32:46                                                    connect_as_sta.py:180DEBUG    17:26:01.301 Received response at GoProUuid.NETWORK_MANAGEMENT_RSP_UUID: 08:02:84:08:01:10:02:18:14                         decipher_response.py:161DEBUG    17:26:01.309 self.bytes_remaining=0                                                                             ble_command_get_hardware_info.py:100                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  DEBUG    17:26:54.196 Received response at GoProUuid.NETWORK_MANAGEMENT_RSP_UUID: 04:02:0c:08:00                                     decipher_response.py:161DEBUG    17:26:54.210 self.bytes_remaining=0                                                                             ble_command_get_hardware_info.py:100INFO     17:26:54.223 Received network provisioning status: provisioning_state: PROVISIONING_UNKNOWN                                    connect_as_sta.py:191                                                                                                                                                             ERROR    17:26:54.236 RuntimeError('Unexpected provisioning state: 0')                                                                  connect_as_sta.py:229`

I would like to know if this issue can be solved, thanks in advance.

Adrián.

tcamise-gpsw, 2025-03-19:
There seems to be a couple different topics here but here are some notes:
1. No, there is no way to set the camera into pairing mode via an API call. This is purposeful as it is considered a security risk.
2. The tutorials are only designed to work with the latest camera (currently Hero 13). They _should_ still work with Hero 9. Can you please post the entire log?

arodrigoca, 2025-03-23:
Hello,

Yes, you can find the log file attached here.

Regards, thanks you.

[log.txt](https://github.com/user-attachments/files/19410468/log.txt)

tcamise-gpsw, 2025-03-25:
Well for what its worth, I just tested with my Hero 9 with the tutorial 6 script and it is working with my home network.

The error from the log you posted is simply `PROVISIONING_UNKNOWN` which unfortunately does not provide any useful information.

My best guess is that there is something special about the network you are connecting to that the camera does not like. One thing to note is that Hero 9 will not work if the network does not have internet access.