gopro/OpenGoPro #689: pypi: Requires >=3.9

martincarapia, 2025-03-20:
**Component**
- Documentation

**Describe the bug**
Pypi says python 3.9 is supported although I get code errors saying it isn't. 

**To Reproduce**
Steps to reproduce the behavior:
Try and run latest (0.19.3) on python 3.9

**Expected behavior**
```bash
    raise RuntimeError("Python >= 3.10 and < 3.13 must be used")
E   RuntimeError: Python >= 3.10 and < 3.13 must be used
=========================== short test summary info ============================
ERROR tests/test_gopro_manager.py - RuntimeError: Python >= 3.10 and < 3.13 must be used
ERROR tests/test_gopro_manager.py - RuntimeError: Python >= 3.10 and < 3.13 must be used
ERROR tests/test_gopro_manager.py - RuntimeError: Python >= 3.10 and < 3.13 must be used
ERROR tests/test_gopro_manager.py - RuntimeError: Python >= 3.10 and < 3.13 must be used
```

**Screenshots**

[image]

**Hardware**
- Desktop MacOS Sequoia 15.3.2

tcamise-gpsw, 2025-03-21:
Thanks for pointing this out. It is fixed now as of 0.19.4.

tcamise-gpsw, 2025-03-21:
sorry if it wasn't clear...the solution was to update the PyPi metadata to show the correctly supported version (3.10, 3.11, and 3.12)

closed 2025-03-21 by tcamise-gpsw