gopro/OpenGoPro #691: Kotlin SDK Dependency

keshara2032, 2025-03-21:
Hi everyone, I am very new to Android/Kotlin programming and this is me trying to develop an app for our research. I am quite stuck in figuring out how to add the Kotlin dependency to the build gradle. I thought according to https://gopro.github.io/OpenGoPro/kotlin_sdk/ I could just add a dependency like (    implementation("com.gopro.opengopro:kotlin-sdk:<latest-version>")). For Python I easiliy installed the pip library and managed to set it up.
Am I missing something here? This might be a really stupid question but any advice is highly appreciated.

tcamise-gpsw, 2025-03-21:
You're basically correct. The Kotlin equivalent to pick is grade and the equivalent to pypi is Maven.

Unfortunately I haven't cleared all of the red tape yet to get it on Maven. When I do, the instructions to add it as a dependency will be here: https://github.com/gopro/OpenGoPro/tree/main/demos/kotlin/kmp_sdk#setup

keshara2032, 2025-03-21:
Thank you so much!! I really appreciate all this work. So helpful for our work.

grzesiek2010, 2025-04-15:
Hi @tcamise-gpsw,  
Any updates on this? I'd love to be able to use it.
Do you know when we can expect it to be available?

tcamise-gpsw, 2025-04-16:
I'm still unfortunately having to work through legal stuff. I would say "soon", hopefully within a few weeks.

ankush19902, 2026-04-06:
Hey, 
Any update on this, please !!

mprins, 2026-09-16:
Is there any update on when the sdk will be available on maven? (or similar)