gopro/OpenGoPro #693: Unable to get the streaming data from the GoPro 12 connected to the Raspberry Pi 4B.

Jenanaputra, 2025-03-26:
I am using a GoPro Hero 12 Black and have connected it to a Raspberry Pi 4B via USB. I am using Ubuntu 22.04. I tried running the command:

`start-webcam <3-digit serial number> -p 9000  ` in my raspi.

However, it only produced the following log:

```
INFO:root:Starting webcam
DEBUG:root:Sending webcam/start: {'port': 9000}
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 172.22.191.51:8080
```
…and nothing else. It is supposed to stream data via UDP.

When I connected the camera to my laptop and ran the same command, it worked.

Does anyone know how to solve this? Thanks!

tcamise-gpsw, 2025-03-26:
To be clear, you are talking about the [multi webcam demos](https://github.com/gopro/OpenGoPro/tree/main/demos/python/multi_webcam). 

First, the bad news...I never actually tested this on any RPI. I just went to do it now and the package doesn't even install since there is no numpy distribution for my RPI.  So I'm going to note in the docs that it is not supported on RPI.

Now some good news....this does work on my RPI with the Open GoPro [Python SDK](https://gopro.github.io/OpenGoPro/python_sdk/). I tested by running:
`gopro-photo --wired` and it automatically discovered my gopro and took a picture. So I would recommend using that

Jenanaputra, 2025-03-27:
Is your RPI operating system ubuntu ? I run `gopro-photo --wired` I got this log :
```
INFO     11:55:22.276 Querying mDNS to find _gopro-web._tcp.local....                                                                                                                            mdns_scanner.py:69
ERROR    11:55:32.303 FailedToFindDevice('A scan timed out without finding a device')
```

to make sure the gopro detected in my RPI, i did `lsusb` and got this :
`Bus 001 Device 003: ID 2672:0059 GoPro HERO12 Black`

which means it is connected.

Jenanaputra, 2025-03-29:
Here is my system info :

- RPI : RPI 4 B
- RPI OS :

```
PRETTY_NAME="Ubuntu 22.04.5 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.5 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
```

- camera : Gopro hero 12 Black
- open-gopro version : 0.19.4

1. When I tried to do `mdns-scan` with gopro connected to RPI, I only got the "Browsing ..." respond and it took a lot of time.
2. I did `curl http://172.22.191.51:8080/gopro/version` with gopro connected to RPI, nothing came up

tcamise-gpsw, 2025-04-01:
If your OS can not find the camera, then there is nothing the Python demos can do. I don't know enough about Linux, USB, etc to comment on why your OS can not find it.

Jenanaputra, 2025-04-02:
Thanks!, I'll try to install the raspian and then try to run/use the sdk.

closed 2025-04-03 by Jenanaputra