gopro/OpenGoPro #714: Win11 drops wifi connection when exiting async with WirelessGoPro

Brumbay, 2025-03-30:
**Component**
What is the bug in?
Python SDK

**Describe the bug**
When exiting the async with WirelessGoPro block, Windows 11 also drops the wifi connection on the wifi interface that was used.

**To Reproduce**
Steps to reproduce the behavior:
1. Open sdk_wireless_camera_control -> open_gopro -> demos -> cohn.py
2. Place debug point at last line of with block
3. See that when we get out of it the wifi connection of the system is dropped

**Expected behavior**
Home wifi connection should not be dropped

**Screenshots**

**Hardware**
- Camera: Hero12
- Desktop if applicable Windows 11

tcamise-gpsw, 2025-04-01:
This is the current intended behavior. If you want to keep the `WirelessGoPro` object alive, you can manually use [open / close](https://gopro.github.io/OpenGoPro/python_sdk/usage.html#wireless-opening) instead of the context manager.

tcamise-gpsw, 2025-04-02:
I haven't tested on Windows 11 (only Windows 10) but I have seen anecdotally that it is working.

Anyway I'm not able to reproduce the issue you are reporting. I run [gopro-cohn demo](https://github.com/gopro/OpenGoPro/blob/main/demos/python/sdk_wireless_camera_control/open_gopro/demos/cohn.py), let it provision, and exit. Then I can continuously use the recommended curl command to query the camera state.

Can you post the `gopro_demo.log` that the demo is producing?

closed 2025-05-13 by github-actions[bot]