gopro/OpenGoPro #724: Python SDK: create WirelessGoPro() from COHN data

Brumbay, 2025-04-18:
Hello!

I'm using the Python SDK to communicate with A GoPro hero 12 over COHN.

My problem is that if I restart the application, I don't know how to get a WirelessGoPro object or something similar from the cohn data I have. I haven't found anything on the docs nor in the codebase.

Do I have to just go back to using CURL commands?

tcamise-gpsw, 2025-04-18:
Hello. I'm actually working right now (internally)  on a PR to implement creating `WirelessGoPro` from a COHN database that is managed by the SDK. It should hopefully be pushed next week.

Brumbay, 2025-04-18:
Well that's really lucky for me then! Thanks

tcamise-gpsw, 2025-04-21:
I'm going to re-open this in case I end up getting sidetracked from completing this

closed 2025-05-12 by tcamise-gpsw

tcamise-gpsw, 2025-05-12:
I just pushed a major update (0.20.0) that adds this functionality. The two big changes that matter for this ticket are:
- [WirelessGoPro](https://gopro.github.io/OpenGoPro/python_sdk/api.html#wireless) initialization arguments
   - `interfaces` can be set to only connect to COHN
   - `cohn_db` can be used to set the COHN database path. Its optional and will just default to `cohn_db.json` if you don't pass it.

The idea here is that the `WirelessGoPro` will track the COHN credentials as they are received and store them in the database.

Also, COHN functionality has now been abstracted into a [CohnFeature](https://gopro.github.io/OpenGoPro/python_sdk/api.html#open_gopro.features.cohn_feature.CohnFeature). Check out the [cohn demo](https://github.com/gopro/OpenGoPro/blob/main/demos/python/sdk_wireless_camera_control/open_gopro/demos/cohn.py) script, specifically its improved help, to demonstrate the usage.