gopro/OpenGoPro #734: GoPro 13 Black COHN with no DHCP network

ramzesko, 2025-04-28:
We are building an app using multiple GoPro 13 Black cameras to record synchronously, stop and download records based on external signal.
We concluded that using COHN mode would be the best to communicate multiple cameras at the same time. Unfortunately the network where we can communicate the app and cameras together with the trigger signal is a kind of closed industrial network with manual IP assignation and no DHCP server.
I've modified a piece of code in connect_sta.py (tutorial module 6 - python)
`connect_request = bytearray(
                [
                    0x02,  # Feature ID
                    0x05,  # Action ID
                    *proto.RequestConnectNew(ssid=entry.ssid,
                                             password=password,
                                             ).SerializePartialToString(),
                ]
            )`
TO:
`connect_request = bytearray(
                [
                    0x02,  # Feature ID
                    0x05,  # Action ID
                    *proto.RequestConnectNew(ssid=entry.ssid,
                                             password=password,
                                             static_ip=socket.inet_aton(ip_address),
                                             gateway=socket.inet_aton(gateway),
                                             subnet=socket.inet_aton(subnet),
                                             dns_primary=socket.inet_aton(dns_primary),
                                             dns_secondary=socket.inet_aton(dns_secondary)
                                             ).SerializePartialToString(),
                ]
            )`

That way I was able to pass the network settings and get provisioned but I only went into the "COHN_STATE_ConnectingToNetwork" status few times finally without success.
Many times I'm also getting a provisioning state: PROVISIONING_ERROR_FAILED_TO_ASSOCIATE
RuntimeError('Unexpected provisioning state: 7')

It would be great to know if my approach is right or if the hardware is able to connect such network at all.

tcamise-gpsw, 2025-04-28:
Hello. There is a new `bypass_eula_check` argument to the [RequestConnectNew ](https://gopro.github.io/OpenGoPro/ble/protocol/protobuf.html#requestconnectnew) protobuf message that allows connecting to networks without internet access. 

I have no idea what the code you pasted above is doing. I guess those are arguments to `requests`? Anyway you shouldn't need any of that if you set `bypass_eula_check` to True.

jszlavik-mal, 2025-05-06:
Hello! I am also using a wifi network with no internet access. I have tried your parameter and matched the code, but the gopro refuses to connect and crashes out. It will connect to a network with internet and my phone can connect just fine to this same internetless network. Has this parameter been verified or are there any other components that need to be implemented with this?

tcamise-gpsw, 2025-05-06:
Yes I tested it and it works. Please:
- ensure you are using Hero 13 as that it is the only place the parameter will work
- post your parsed log of the failure

tcamise-gpsw, 2025-05-08:
Closing this since it has been implemented and @jszlavik-mal 's issue is being separately tracked in #744

closed 2025-05-08 by tcamise-gpsw