gopro/OpenGoPro #738: Requested update to util.py ainput()

ksaylor, 2025-04-30:
### Description
I am encountering an issue when building an executable using cx_freeze that is not terminal-based. The current implementation of `ainput` in `util.py` leads to an `AttributeError: 'NoneType' object has no attribute 'write'`. To resolve this issue, I am requesting an update to the `ainput` function.

### Current Code
```python
async def ainput(string: str, printer: Callable = sys.stdout.write) -> str:
    """Async version of input

    Args:
        string (str): prompt string
        printer (Callable): Callable to display prompt. Defaults to sys.stdout.write.

    Returns:
        str: Input read from console
    """
    await asyncio.get_event_loop().run_in_executor(None, lambda s=string: printer(s + " "))  # type: ignore
    return await asyncio.get_event_loop().run_in_executor(None, sys.stdin.readline)
````

### Proposed Code Change
```python
async def ainput(string: str, printer: Callable = None) -> str:
    """Async version of input

    Args:
        string (str): prompt string
        printer (Callable): Callable to display prompt. Defaults to sys.stdout.write.

    Returns:
        str: Input read from console
    """
    if not printer:
        printer = sys.stdout.write
    await asyncio.get_event_loop().run_in_executor(None, lambda s=string: printer(s + " "))  # type: ignore
    return await asyncio.get_event_loop().run_in_executor(None, sys.stdin.readline)
````
### Reason for the Update
When running an executable built with the current version of util.py, using cx_freeze for a non-terminal-based application, the following error occurs (this is snippet of the Traceback):

````
in <module> async def ainput(string: str, printer: Callable = sys.stdout.write) -> str:
                                                   ^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'write'
````

This error occurs because `sys.stdout` is `None` in a non-terminal environment, leading to the AttributeError. By modifying the default value for `printer` to None and setting it to `sys.stdout.write` within the function if it is not provided, this issue is resolved.

Thank you for considering this change.

tcamise-gpsw, 2025-04-30:
Hello. I'm kind of confused why you need `ainput` if you're not using a terminal application. But anyway this certainly seems reasonable.

I've linked a PR here to implement this. Can you review?

ksaylor, 2025-04-30:
Hi @tcamise-gpsw as far as I can tell it's because I am using a logger. I actually don't make any calls to `ainput`. Thank you for your quick response and I will review the PR :-)

tcamise-gpsw, 2025-04-30:
This has been merged and released to PyPi in version 0.19.8

ksaylor, 2025-04-30:
@tcamise-gpsw thank you again for addressing this so quickly!