gopro/OpenGoPro #744: COHN Provisioning High Failure Rate

jszlavik-mal, 2025-05-08:
Preface: I am currently using the most up to date version of the opengopro repository and GoPro 13 Blacks with the most up to date software.

Problem: Is there any way to improve the reliability of provisioning gopro cameras onto a network?

I am trying to create a COHN with the cameras. Using the tutorials, the demos, and my own adaption of the opengopro library I notice that connecting a single camera has an extremely and a high failure rate when attempting to connect a camera to the network with no pattern of when it does work.
I have factory reset the cameras, power cycled the cameras, and power cycled the hosting network machine. No individual or combination of the previous actions seemed to give consistent or improved results on provisioning a camera to the network. Any other device can connect to the network with no issues at all.
It appears the GoPro will connect to bluetooth fine. When requesting the camera to scan for networks, very often it fails to respond with anything. When it does find the correct network, it very often has difficulty connecting to the network. This combination leads to a consistent unsuccessful result.
Sometimes it will work back to back then it wont work for 10+ times. Then will work once, then back to not working. No pattern at all. 

Has anyone been able to achieve consistent camera provisioning to a single network? If so, what was the method?

tcamise-gpsw, 2025-05-08:
Hello. Can you please post parsed logs of the failure as requested by the bug template?

In any case, I don't think I've ever seen a failure to scan for, or connect to, a WiFi Access Point using the [Python SDK](https://gopro.github.io/OpenGoPro/python_sdk/). 

Based on your comment [here](https://github.com/gopro/OpenGoPro/issues/734), I assume that you are trying to connect to a network without internet access? Is there anything else notable about this network such as it being [WPA/WPA2](https://github.com/gopro/OpenGoPro/issues/391)?

jszlavik-mal, 2025-05-09:
Hello!

Reference:
I am currently using the command line application demo "gopro-cohn" also located at demos/python/sdk_wireless_camera_control/open_gopro/demps/cohn.py
I am trying to connect to a network with internet access. I start without, but it wasn't working so I moved to one with internet acces just to rule that out (The sdk I was working with hadn't been updated to that parameter at the time, but everything is now up to date.)
The wifi I am using is my iPhones standard hotspot at the moment.

The camera is not connecting, however I am not sure if it because of the provisioning or not. It certainly had successfully connected at least once. I even still have the cohn.crt from it. When line 31(cohn.py) is run to check if the network has been provisioned, requests cohn status and it returns that it has been provisioned, but returns no ip, or ssid (not sure if that is normal). 

[image]

It will then send the same status request with the same result. After entering some log functions, the script is timing out on line 484 (gopro_wireless.py)

[image]

It looks as if it doesn't store/queue the credentials to be pulled from. I have tried to force reprovisioning the network by bypassing the check to see if it already has been provisioned to fix it just in case if it wasn't saved write, but I still have the same issue. The way it is written I believe the username, password, and ip should be returned and used when requesting cohn status, but as you can see from the photo that none of those come back. Is there something wrong with the cameras or am I looking in the wrong direction here?

tcamise-gpsw, 2025-05-09:
Can you post the log? There should be `gopro_demo.log` file from wherever you are calling this.

tcamise-gpsw, 2025-05-09:
The script should be working as-is but I'm also currently [in the middle](https://github.com/gopro/OpenGoPro/issues/724#issuecomment-2815804013) of a major update around COHN functionality to, among other things, add a database of credentials. Let's review your log to make sure nothing else is going on but it might be worth waiting until I push this update.

jszlavik-mal, 2025-05-09:
That is really great to know. I might have another get around for what I am trying to do right now and i can wait for your COHN update. Would love to see what you've done. 
I didn't realize these files were there. Apologies. Thank you

[gopro_demo.log](https://github.com/user-attachments/files/20126976/gopro_demo.log)

tcamise-gpsw, 2025-05-12:
@jszlavik-mal Hey I just pushed a major update (0.20.0) to rework COHN. There are some more comments [here](https://github.com/gopro/OpenGoPro/issues/724#issuecomment-2874419781).

Let me know if you run into any issues and we can address them here.

jtiano76, 2025-05-14:
I'm having similar issues I notice I can only connect COHN with a wifi network that has internet.
I'm wanting to use the COHN feature without internet service.

Any thoughts on how to achieve. Keep getting provisioning error 7 and 11

tcamise-gpsw, 2025-06-18:
I believe this has been fixed with the 0.20.0 changes. Please attach a log here if there are still issues.

ksaylor, 2025-07-16:
First, I want to thank you for all the hard work you’ve put into this, @tcamise-gpsw! I also really like the rework you've done and the shift to a Features-based approach—it makes for a much cleaner interface, in my opinion.

I'm in the process of migrating my codebase from a pre-0.20.0 version to the latest, and I’ve run into some issues using the cohn.py example. Specifically, scan_wifi_networks() intermittently fails to find any networks. I’ve been trying to reproduce the issue consistently to better understand what’s going on. It seems that once a camera fails to find networks, I have to manually reset wireless settings via the camera’s Preferences. However, if the camera does detect networks, the rest of the process completes successfully.

With that in mind, I have a few clarification questions to ensure I understand the COHN flow correctly:

1. COHN Certificate Validity
From what I’ve read, the COHN certificate and the associated username/password are valid for one year—is that correct? Or is it just the certificate that expires? Also, do the certificate and credentials persist if the battery is removed from the camera?

2. gopro.access_point.connect()
This step makes sense to me (aside from the intermittent network scanning issue), so I don't have any specific questions here.

3. gopro.cohn.configure(force_reprovision=True)
This is where I need more clarity. In earlier versions (pre-0.20.0), I only had to provision the camera if power (battery or external) had been removed. Otherwise, provisioning was persistent. It looks like a new certificate (and possibly a new password?) is generated when the camera is reprovisioned.
So my questions are:

Should I generally avoid forcing reprovision unless necessary?

Does the cert/username/password persist even after a power loss?

In other words, will gopro.cohn.is_configured still return True after the camera has been powered off?

Thanks in advance for any guidance you can provide!

closed 2025-08-28 by github-actions[bot]