gopro/OpenGoPro #759: KMP SDK does not allow using HTTP APIs when using COHN

TinfoilSubmarine, 2025-05-21:
**Component**
KMP SDK

**Describe the bug**
Using commands like startPreviewStream, startWebcam, etc. fail with error:
```
com.gopro.open_gopro.operations.OperationUnsupportedForCommunicator: Start Preview Stream not supported on HTTP
```

**To Reproduce**
Steps to reproduce the behavior:
1. Take GoPro through COHN setup steps (connect to access point, provision COHN).
2. Try to use `goPro.commands.startPreviewStream()`

**Expected behavior**
KMP SDK should allow using commands that communicate over HTTP when a COHN connection is available.

**Logs**
Error provided above.

**Screenshots**
n/a

**Hardware**
- Camera: Hero13
- Smartphone: Pixel 5

**Additional context**
n/a

tcamise-gpsw, 2025-06-20:
I think the problem here is that you haven't actually "connected" COHN. When COHN provisioning is done via BLE, the camera is simply ready to accept a COHN connection. You still need to actually call the [connect](https://gopro.github.io/OpenGoPro/kotlin_sdk/-open%20-go-pro%20-s-d-k/com.gopro.open_gopro/-ogp-sdk/index.html#-1475456515%2FFunctions%2F1280103317) to open the COHN channel.

In the demo app, this can be done by going back to the camera chooser, scanning via DNS, and selecting the WiFi device.

I think this is actually fine behavior for the demo since its clearly demonstrating the SDK functionality but if you feel strongly about it feel free to make a PR to update the demo.

closed 2025-06-20 by tcamise-gpsw

TinfoilSubmarine, 2025-07-09:
Thank you, this was just a bit confusing when using the SDK. I've got it working now following your guidance.