gopro/OpenGoPro #764: COHN is provisioned but error SSL: WRONG_VERSION_NUMBER

Brumbay, 2025-05-27:
**Component**
What is the bug in?
- SDKs (Python / Kotlin)
- Demos (Python)

**Describe the bug**
After cohn is provisioned, the test request fails with error:

SSLError(MaxRetryError("HTTPSConnectionPool(host='192.168.178.90', port=8080): Max retries exceeded with url: /gopro/camera/state (Caused by SSLError(SSLError(1, '[SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:1000)')))"))

Example curl command DOES work. Only the python code can't connect.

**To Reproduce**
Steps to reproduce the behavior:
1. Execute command gopro-cohn --ssid wifissid --password wifipassword

Alternatively, the same happens with 
gopro-cohn --identifier id

**Expected behavior**
Test communication should run

**Logs**
Logs attached

[gopro_demo.log](https://github.com/user-attachments/files/20466700/gopro_demo.log)

**Hardware**
- Camera: Hero12 (running 2.32.0
- Desktop if applicable: Linux

tcamise-gpsw, 2025-05-30:
I just released a critical bug fix in open-gopro 0.20.2. Can you test again with that version?

Brumbay, 2025-05-31:
I confirm it's now working as expected! Thanks @tcamise-gpsw!