gopro/OpenGoPro #776: Connecting to access points via the GoPro compose app rarely appears to work

MRL-00, 2025-06-12:
**Component**
What is the bug in?
- Compose App - Kotlin
- Specs (BLE, HTTP)

**Describe the bug**
Using the sample compose app from Open GoPro, trying to connect to an access point doesn't always work. If I connect to `network a`, it works. Then I try connect to `network b` and it just hangs on the "connecting" state.  Then closing and reopening the app causes the connection to `network a` to hang.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to compose app repository, clone and wait for gradle
2. Connect to a camera via BLE
3. Go to `Connect to access point` and scan for SSIDs
4. Connect to any ssid, it should work. 
5. Go back to the previous screen and then repeat steps 3 and 4, this time selecting a different access point.
6. It will now just stay as "connecting to `network name`. It will never complete.

**Expected behavior**
Should be able to gracefully disconnect from one WiFi network and connect to another WiFi network. 

**Screenshots**
[image]

**Hardware**
- Camera: Hero13.
- OnePlus 13

MRL-00, 2025-06-16:
Some further logs captured
```
 Keep Alive finished with result ==> Success: 
2025-06-16 14:25:19.521 15989-16169 BleCommunicator         gopro.open_gopro.compose_app         E  BLE Write / Receive Notification timeout of 15s
2025-06-16 14:25:19.521 15989-16169 BleCommunicator         gopro.open_gopro.compose_app         D  5. Checking received response Set Setting Response: LED vs expected Set Setting Response: LED
2025-06-16 14:25:19.522 15989-16169 Keep Alive              gopro.open_gopro.compose_app         I  Keep Alive finished with result ==> com.gopro.open_gopro.BleError: BLE Write / Receive Notification timeout of 15s
2025-06-16 14:25:19.522 15989-16169 GoPro 7983              gopro.open_gopro.compose_app         W  Failed to send keep alive.
2025-06-16 14:25:20.796 15989-16169 Keep Alive              gopro.open_gopro.compose_app         I  Executing Keep Alive with BLE communicator
2025-06-16 14:25:20.798 15989-16041 BleCommunicator         gopro.open_gopro.compose_app         D  1. Writing to CQ_SETTINGS ==> 5b 01 42
2025-06-16 14:25:20.799 15989-16041 Kable/Peripheral        gopro.open_gopro.compose_app         D  FE:8B:9C:3A:17:FC write
                                                                                                    service: 0000fea6-0000-1000-8000-00805f9b34fb
                                                                                                    characteristic: b5f90074-aa8d-11e3-9046-0002a5d5c51b
                                                                                                    writeType: WithResponse
                                                                                                    data: 20 03 5B 01 42
2025-06-16 14:25:20.820 15989-16157 Kable/Callback          gopro.open_gopro.compose_app         D  FE:8B:9C:3A:17:FC onCharacteristicWrite
                                                                                                    service: 0000fea6-0000-1000-8000-00805f9b34fb
                                                                                                    characteristic: b5f90074-aa8d-11e3-9046-0002a5d5c51b
                                                                                                    status: GATT_SUCCESS(0)
2025-06-16 14:25:20.821 15989-16041 BleCommunicator         gopro.open_gopro.compose_app         D  2. Waiting to receive response Set Setting Response: LED
2025-06-16 14:25:20.859 15989-16157 Kable/Callback          gopro.open_gopro.compose_app         D  FE:8B:9C:3A:17:FC onCharacteristicChanged
                                                                                                    service: 0000fea6-0000-1000-8000-00805f9b34fb
                                                                                                    characteristic: b5f90075-aa8d-11e3-9046-0002a5d5c51b
                                                                                                    data: 02 5B 00
2025-06-16 14:25:20.861 15989-16041 KableDevice             gopro.open_gopro.compose_app         D  Received notification 02 5b 00 on b5f90075-aa8d-11e3-9046-0002a5d5c51b
2025-06-16 14:25:20.863 15989-16161 BleCommunicator         gopro.open_gopro.compose_app         D  3. Received notification on CQ_SETTINGS_RESP: 02 5b 00
2025-06-16 14:25:20.863 15989-16161 BleCommunicator         gopro.open_gopro.compose_app         D  4. Emitting accumulated response Set Setting Response: LED
2025-06-16 14:25:20.865 15989-16038 BleCommunicator         gopro.open_gopro.compose_app         D  5. Checking received response Set Setting Response: LED vs expected Set Setting Response: LED
2025-06-16 14:25:20.867 15989-16169 BleCommunicator         gopro.open_gopro.compose_app         D  6. Returning accumulated response Set Setting Response: LED
2025-06-16 14:25:20.868 15989-16169 Keep Alive              gopro.open_gopro.compose_app         I  Keep Alive finished with result ==> Success: 
2025-06-16 14:25:22.523 15989-16169 Keep Alive              gopro.open_gopro.compose_app         I  Executing Keep Alive with BLE communicator
2025-06-16 14:25:22.525 15989-16160 BleCommunicator         gopro.open_gopro.compose_app         D  1. Writing to CQ_SETTINGS ==> 5b 01 42
2025-06-16 14:25:22.525 15989-16160 Kable/Peripheral        gopro.open_gopro.compose_app         D  CB:AD:33:15:0E:70 write
                                                                                                    service: 0000fea6-0000-1000-8000-00805f9b34fb
                                                                                                    characteristic: b5f90074-aa8d-11e3-9046-0002a5d5c51b
                                                                                                    writeType: WithResponse
                                                                                                    data: 20 03 5B 01 42
2025-06-16 14:25:23.308 15989-16628 Kable/Callback          gopro.open_gopro.compose_app         D  CB:AD:33:15:0E:70 onCharacteristicWrite
                                                                                                    service: 0000fea6-0000-1000-8000-00805f9b34fb
                                                                                                    characteristic: b5f90074-aa8d-11e3-9046-0002a5d5c51b
                                                                                                    status: GATT_ERROR(133)
[... cut ...]
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1005)
                                                                                                    	Suppressed: kotlinx.coroutines.internal.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@f04940b, Dispatchers.Main.immediate]
2025-06-16 14:25:23.425 15989-15989 Process                 gopro.open_gopro.compose_app         I  Sending signal. PID: 15989 SIG: 9
```

MRL-00, 2025-06-18:
Thanks @tcamise-gpsw. In terms of replicating this, the easiest way is to join a network, disconnect, then join a different network. That seems to cause it for me.

MRL-00, 2025-06-18:
Cheers @tcamise-gpsw. Is this just a change in the Kotlin SDK or will this require some firmware updates to the GoPro device as well? We're not specifically using the Kotlin SDK but have had similar results from our codebase as what was being displayed in the demo app.

tcamise-gpsw, 2025-06-18:
it's just in the SDK. Basically you need to ensure you disconnect before connecting.

The disconnect command is really just disabling station mode: https://gopro.github.io/OpenGoPro/ble/features/access_points.html#disconnect-from-access-point

MRL-00, 2025-06-18:
Thanks. We'll do some testing on our end and will report back.